AndrejOrsula/drl_grasping #89: drl_grasping

lisheng-han, 2021-07-30:
Andrej ! 
When I follow your steps——according to the docker/Dockerfile

colcon build --merge-install --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

I don't know how to solve this problem---When I carry out the above command

The output is <RuntimeError: The package_dir contains a mapping from 'ocnn' to 'ocnn' which is also a key>
Summary: 0 packages finished [9.43s]
  1 package failed: ocnn
  7 packages aborted: Caffe Octree drl_grasping iDynTree ign_moveit2 ros_ign_bridge ros_ign_gazebo
  4 packages had stderr output: Caffe iDynTree ocnn ros_ign_bridge
  9 packages not processed

closed 2021-07-30 by lisheng-han

AndrejOrsula, 2021-07-30:
Hello @lisheng-han.

Have you solved the error? If not, I have some suggestions (either of them should fix this issue):

1) `cd` into `drl_grasping` directory if you followed [docker/Dockerfile](https://github.com/AndrejOrsula/drl_grasping/blob/3d9fa038dc2855b9ca97d9d1570b06abe17a0b89/docker/Dockerfile) (including directory structure).
2) Move `ocnn` outside of the workspace that contains ROS 2 repositories (`drl_grasping`, `ign_moveit2`, ...).
3) Add `COLCON_IGNORE` file inside the low-level directory that contains `ocnn` (e.g. `touch ./O-CNN/COLCON_IGNORE`).

This is because `ocnn` dependency cannot be built/installed directly with `colcon`. In fact, most of the used dependencies cannot.

lisheng-han, 2021-07-30:
Okay, Andrej ! Thanks for your advice，I will reconfigure the environment