AndrejOrsula/drl_grasping #97: Kinova arm with SAC Algorithm not learning

dsriaditya999, 2022-02-02:
Hi,

I have installed the docker image and I have tried to run `ros2 run ex_train.bash` with the default configuration for Kinova arm and SAC algorithm for 400k timesteps on my laptop with dedicated GPU. But the agent does not seem to be learning to grasp. Any suggestions would be really helpful @AndrejOrsula.

[image]

[image]

I am currently running the train using the TQC algorithm.

AndrejOrsula, 2022-02-02:
Hello,

Unfortunately, it is difficult to debug these kind of issues. It is unexpected, so I think it might related to the simulation or data transfer in one way or another. I can give you suggestions on what I would try next and see if we can resolve this. If there are some warnings/error, that would also help the process.

1) Run the default pre-trained agent to see if it gives any successful grasps. If not, the issue originates with the host setup.
2) Open Ignition Gazebo client or RViz2 and investigate if you can see any visuals at all.
3) If not and you use some custom locale, trying `export LC_ALL=C` or `export LC_ALL=en_GB.utf8` won't hurt and it fixes one of the infamous problem with parsing that Ignition currently has/had.
4) If that does not help, make sure you get no errors regarding ROS 2 communication, e.g. network interface not supporting multicast.
5) Use `panda` robot as I did most tests with that one (before moving to the robot you need). I actually just realised that the training example defaults to Kinova. This is actually a mistake I did not notice before. It should default to `panda`.
https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/examples/ex_train.bash#L17-L19

If you continue experiencing this problem, I will try to reproduce it. Unfortunately, I won't really have any time or resources to do that in these months. And at that point, I am planning to (hopefully) release a refactored version of this repository anyway.

---

Regarding Kinova arm -- As I already mentioned in https://github.com/AndrejOrsula/drl_grasping/issues/96#issuecomment-1025857288, `j2s7s300` is not really configured well for this project yet. This includes many things:
- Collision geometry was not remodelled manually for this robot. It is way slower than it needs to be (too much geometry), and I have seen the gripper clipping through objects in some instances (where the object might get stuck to the gripper).
- PID tuning is quite bad in this version as far as I can remember, especially for the gripper.
- Lack of testing. I only trained 2 or 3 of agents for `j2s7s300` with TQC to make sure it works. I can confirm that it started to learn with default configuration within 100k steps. I don't think I tested on using Docker though, only local installation.
- Incorrect kinematic configuration is being used for the 3-finger gripper. This was recently fixed for `kinova-ros` repo, but it's not updated here yet.

dsriaditya999, 2022-02-02:
Thank you for your response. I have tried to run the default pre-trained agent and I have seen many successful grasps. I am now training the Kinova arm with TQC algorithm (I hope it works!). I would also try to run the train experiment on panda arm as you suggested just to confirm everything is ok. 

---

I am really motivated and interested in your work. I am actually working with a Kinova arm in my lab and I wanted to configure this project so that I can do the experiments in simulation and with the real-robot as you have mentioned in the Thesis. So I wanted to ask if you could give me some pointers so that I can work on accomplishing this?

AndrejOrsula, 2022-02-02:
For sim-to-real evaluation, there is this commit https://github.com/AndrejOrsula/drl_grasping/commit/aa7d75058c8189e3e876d84b95f58450c72976b2 that contains some very basic and hacky realtime runtime and implementation for UR and Panda (It was not designed properly, but made for a quick evaluation). It is probably very messy too, so I wouldn't actually use it anymore. If you really want to adapt it for Kinova arm, then using their internal controllers and official API is probably the easiest.

For the next release that I keep mentioning, I will add a bit better interface for sim-to-real transfer and use MoveIt 2 via `ros2_control` (or `ros_control` via `ros1_bridge` if robot is not yet supported for ROS 2) for every robot instead of a custom interface as I did for the quick test above.

AndrejOrsula, 2022-07-31:
Closed due to inactivity...

closed 2022-07-31 by AndrejOrsula