AndrejOrsula/drl_grasping #98: Error while running training bash script

Shanki5, 2022-02-27:
Two errors arise @AndrejOrsula 
1. No module name 'ocnn' : I have built the octree fork and added pwd to path. Still library is not found
2. ign_moveit2 is not found. The repo seems to be renamed as ign_moveit2_examples. How do I change that?

AndrejOrsula, 2022-02-27:
Hello @Shanki5,

> 2) `ign_moveit2` is not found. The repo seems to be renamed as `ign_moveit2_examples`. How do I change that?

This repository is not up-to-date with its dependencies. Fixing all of it would take a lot of your time (ROS 2 side), so I wouldn't recommend you trying that. Please, use [Docker](https://hub.docker.com/repository/docker/andrejorsula/drl_grasping) if you want to experiment with this repository (until it gets updated). Instructions are in README.

> 1) No module name 'ocnn' : I have built the octree fork and added pwd to path. Still library is not found

This is related to `ocnn` repository and the way in which it was installed, so I am not sure what would be causing with the information you provided. If you follow Dockerfile instructions, it should work without any issues.

AndrejOrsula, 2022-07-31:
Closed due to inactivity...

closed 2022-07-31 by AndrejOrsula