AndrejOrsula/drl_grasping #99: Visualizing

Nils-ChristianIseke, 2022-06-11:
I am trying to visualize the `Reach-Gazebo-v0` task, but i am not succeeding.

Steps to reproduce:
1. Set ENV_ID, ROBOT_MODEL and ALGO inside ex_train.bash to the following values
```
ENV_ID="Reach-Gazebo-v0"
ROBOT_MODEL="panda"
ALGO="tqc"
```
2. ros2 run drl_grasping ex_train.bash
3. ign gazebo -g

The visualization works for `ENV_ID="Grasp-OctreeWithColor-Gazebo-v0"` but not for any of the Reach tasks. 
In Gazebo there is only a blank white world shown. 

This is the output in the terminal after executing ros2 run drl_grasping ex_train.bash:
```
 ros2 run drl_grasping ex_train.bash 
Launching ign_moveit2 in background:
ros2 launch drl_grasping ign_moveit2_headless.launch.py

Executing train command:
ros2 run drl_grasping train.py --env Reach-Gazebo-v0 --algo tqc --seed 42 --log-folder training/Reach-Gazebo-v0/logs --tensorboard-log training/Reach-Gazebo-v0/tensorboard_logs --eval-freq -1 --eval-episodes 10 --env-kwargs robot_model:"panda" 

[INFO] [launch]: All log files can be found below /root/.ros/log/2022-06-11-16-30-31-095293-nils-ThinkPad-X390-Yoga-301513
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [301545]
[INFO] [parameter_bridge-2]: process started with pid [301547]
[INFO] [parameter_bridge-3]: process started with pid [301549]
[INFO] [parameter_bridge-4]: process started with pid [301551]
[INFO] [parameter_bridge-5]: process started with pid [301553]
[INFO] [parameter_bridge-6]: process started with pid [301555]
[parameter_bridge-3] 1654957831.754419 [69] parameter_: selected interface "lo" is not multicast-capable: disabling multicast
[parameter_bridge-2] 1654957831.755151 [69] parameter_: selected interface "lo" is not multicast-capable: disabling multicast
[parameter_bridge-4] 1654957831.759241 [69] parameter_: selected interface "lo" is not multicast-capable: disabling multicast
[parameter_bridge-5] 1654957831.764868 [69] parameter_: selected interface "lo" is not multicast-capable: disabling multicast
[move_group-1] 1654957831.764872 [69] move_group: selected interface "lo" is not multicast-capable: disabling multicast
[parameter_bridge-6] 1654957831.768739 [69] parameter_: selected interface "lo" is not multicast-capable: disabling multicast
[move_group-1] Link panda_link1 had 1 children
[move_group-1] Link panda_link2 had 1 children
[move_group-1] Link panda_link3 had 1 children
[move_group-1] Link panda_link4 had 1 children
[move_group-1] Link panda_link5 had 1 children
[move_group-1] Link panda_link6 had 1 children
[move_group-1] Link panda_link7 had 1 children
[move_group-1] Link panda_hand had 3 children
[move_group-1] Link panda_link8 had 0 children
[move_group-1] Link panda_leftfinger had 0 children
[move_group-1] Link panda_rightfinger had 0 children
[move_group-1] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-1] Loading 'move_group/ClearOctomapService'...
[move_group-1] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-1] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-1] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-1] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-1] Loading 'move_group/MoveGroupMoveAction'...
[move_group-1] Loading 'move_group/MoveGroupPlanService'...
[move_group-1] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-1] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-1] 
[move_group-1] You can start planning now!
[move_group-1] 
========== Reach-Gazebo-v0 ==========
Seed: 42
OrderedDict([('batch_size', 32),
             ('buffer_size', 25000),
             ('ent_coef', 'auto_0.1'),
             ('env_wrapper',
              [{'drl_grasping.envs.randomizers.ManipulationGazeboEnvRandomizer': {'robot_random_joint_positions': True,
                                                                                  'robot_random_joint_positions_std': 0.1}}]),
             ('gamma', 0.95),
             ('gradient_steps', 1),
             ('learning_rate', 'lin_0.0003'),
             ('learning_starts', 0),
             ('n_timesteps', 50000),
             ('noise_std', 0.025),
             ('noise_type', 'normal'),
             ('optimize_memory_usage', True),
             ('policy', 'MlpPolicy'),
             ('policy_kwargs',
              {'n_critics': 2, 'n_quantiles': 25, 'net_arch': [128, 64]}),
             ('target_entropy', 'auto'),
             ('tau', 0.01),
             ('top_quantiles_to_drop_per_net', 2),
             ('train_freq', 1)])
Using 1 environments
1654957834.844785 [69]    python3: selected interface "lo" is not multicast-capable: disabling multicast
[INFO] [1654957835.094510306] [ign_moveit2_py_0]: ign_moveit2_py initialised successfuly
Applying normal noise with std 0.025
Using cuda device
Log path: training/Reach-Gazebo-v0/logs/tqc/Reach-Gazebo-v0_8
Inserting robot
Warning [GenericJoint.hpp:1480] [GenericJoint::setRestPosition] Value of _q0 [0], is out of the limit range [-3.07178, -0.0698132] for index [0] of Joint [Joint].
Inserting default object
Logging to training/Reach-Gazebo-v0/tensorboard_logs/Reach-Gazebo-v0/TQC_8

```

AndrejOrsula, 2022-06-12:
Hello,

Thank you for opening this issue. Yes, I can indeed reproduce your findings with version `1.1.0` and the behavior is unintentional.

One workaround is to manually enable rendering for the environment via `env.render("human")`. You can do that in multiple places, e.g. in `train.py` as  `model._env.render("human")`. Note that this will open the GUI immediately. You should be able to close it and re-open it manually with `ign gazebo -g`, but I can't guarantee that.

I attempted to come up with proper fix but I was not able to make it work for the `Reach*` tasks with the current public version. I believe it has something to do with timing of broadcasting the model insertion, as there are no differences among tasks when it comes to server state broadcasting initialization (the simulation state seems to be broadcast, but without any model information).

This issue won't be in the next release, which should hopefully come soon. Apologies for the inconvenience.

---

Btw, I recommend enabling multicast on your loopback interface via `sudo ifconfig lo multicast` to avoid certain issues that might occur.
>```bash
> 1654957834.844785 [69]    python3: selected interface "lo" is not multicast-capable: disabling multicast
>```

Nils-ChristianIseke, 2022-06-12:
@AndrejOrsula Awesome, thank you for the fast anwser! Just a small correction: 
For me adding `model.env.render("human")` instead of 'model._env.render("human") to the train.py script worked.

closed 2022-06-13 by Nils-ChristianIseke