AndrejOrsula/drl_grasping #100: Adding static Objects to the Reach Task

Nils-ChristianIseke, 2022-06-12:
Hey, i am trying to modify the Reach tasks by adding random static objects as Obstacles. Are there any tips of how to add those? #

AndrejOrsula, 2022-06-13:
You can try taking a look at insertion of objects inside [drl_grasping/envs/randomizers/manipulation.py](https://github.com/AndrejOrsula/drl_grasping/blob/1.1.0/drl_grasping/envs/randomizers/manipulation.py) (e.g. [randomize_object_models()](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/envs/randomizers/manipulation.py#L574) or [randomize_object_primitives()](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/envs/randomizers/manipulation.py#L606)). I don't fully remember the state of code at version `1.1.0`, so it might not be super modular... but you will see.

The list of currently available models (including objects) is inside [drl_grasping/envs/models](https://github.com/AndrejOrsula/drl_grasping/tree/1.1.0/drl_grasping/envs/models) directory. You can add custom ones if you need (by using any of the current models as a template), or just use [drl_grasping/envs/models/random_object.py](https://github.com/AndrejOrsula/drl_grasping/blob/1.1.0/drl_grasping/envs/models/random_object.py) with a specific dataset of SDF models (from Fuel or local Fuel cache).

If you add objects, visual observations should be usable straight away (colour and depth image obs). If you want to utilise state-based observations (vanilla Reach task), then you probably want to add these obstacles into the observation space somehow ([here](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/envs/tasks/reach/reach.py#L73-L81) and [here](https://github.com/AndrejOrsula/drl_grasping/blob/2a58185048328d8f7f0a9b7e1f743e77e73863e9/drl_grasping/envs/tasks/reach/reach.py#L100-L114)). I should note that I have not experimented a lot with the Reach task, as I only used it for quick sanity tests during the early development. It was much simpler to solve this task (get decent results) compared to the Grasp task though (quite simple to solve since actions are in Cartesian space).

Nils-ChristianIseke, 2022-06-13:
Thanks for your tips. :)

closed 2022-06-13 by Nils-ChristianIseke