AndrejOrsula/drl_grasping #103: Severe Warnings due to Namespace Collision in RVIZ

lucarei, 2023-03-27:
Hi Andrej. Thanks for your work and for a previous answer. I am trying to make a local installation (yet tried Docker, it works perfectly but i need much more personalization; for example i want to use the UR5 and a modified environment).
I installed all the needed dependancies following the instructions found in Dockerfile (even colcon command seems to finish correctly).
However, when i run an example (like "ros2 run drl_grasping ex_random_agent.bash") RVIZ gives me the following "severe warning for a namespace collision" and Gazebo is not opened.
[image]
[image]
PS: are all the commands from the Dockerfile enough to install correctly or can you share with me other tips?

lucarei, 2023-03-28:
Issue solved using “export PYTHONPATH=~/ws/src/OCNN/octree/build/python:${PYTHONPATH}

closed 2023-03-28 by lucarei

AndrejOrsula, 2023-03-29:
Glad you were able to resolve it!

> PS: are all the commands from the Dockerfile enough to install correctly or can you share with me other tips?

The Dockerfile "was" enough to install everything correctly. However, the versions of dependencies for some ROS packages and all Gazebo libraries are not fixed (these dependencies are installed as Debian packages that do not archive past releases, so it is not possible to fix them). Therefore, if these introduce any breaking changes (ideally, this should not occur) - the Dockerfile might not be valid anymore, and some issues might appear (e.g. compile-time errors, run-time errors or change in computational/learning performance). ROS 2 `galactic` is now EOL, so it won't be getting updates - which means that these dependencies *should* be fine. However, updates to `fortress` could possibly introduce changes that affect this repository.