AndrejOrsula/drl_grasping #111: Deploying on Franka Research 3 (real)

lif314, 2023-10-27:
Could you please provide a code example for deploying on FR3 using ROS2? Thank you!

AndrejOrsula, 2023-10-27:
Hello,

You can find a general example for evaluating models on real robots under [examples/ex_evaluate_real.bash](https://github.com/AndrejOrsula/drl_grasping/blob/b2283544d19c3b202b14caa03e233ac4598ccee2/examples/ex_evaluate_real.bash). It contains the information about relevant parts that need to be running alongside their configuration.

However, there is no example of using Franka Research 3 specifically. I don't have access to such robot.
You would need to add support for this robot yourself (e.g. in your own fork).

lif314, 2023-10-27:
> Hello,
>
> You can find a general example for evaluating models on real robots under [...]

Thanks for your prompt reply!

closed 2023-10-27 by lif314