AndrejOrsula/drl_grasping #113: gmake: *** [Makefile:146: all] Error 2 Failed   &lt;&lt;&lt; diff_drive_controller [11.9s, exited with code 2]

xjtu19, 2024-12-16:
[image]

xjtu19, 2024-12-16:
Option B – Local Installation
# Build
colcon build --merge-install --symlink-install --cmake-args "-DCMAKE_BUILD_TYPE=Release"

AndrejOrsula, 2024-12-16:
Hello @xjtu19,

The error shown above is from the `ros2_controllers` package used as a dependency in this project. As indicated in [README](https://github.com/AndrejOrsula/drl_grasping?tab=readme-ov-file#overview), this project is based on ROS Galactic. Therefore, it is not expected to be functional with Humble, and there is no plan to support this project and its dependencies beyond Galactic.

If you want to try this project, I would suggest starting with the prebuilt Docker image. That is by far the simplest and most reproducible approach.