AndrejOrsula/drl_grasping #118: Action space, State space, Reward function Locations

naa1824, 2025-02-18:
Hello @AndrejOrsula 

sorry to bother you again,

I was studying the project in the last 2 weeks, and I was trying to find the Action space, State space, and Reward function to understand how they work so I can create my own.

But I am getting lost with files,

could you guide me with this?
which files contain these spaces.. ?

I found the class which create the space, but I didn't understand it

is this the action space? env->task->grasp->grasp.py

```
    def set_action(self, action: Action):

        if self.__preload_replay_buffer:
            action = self._demonstrate_action()

        self.get_logger().debug(f"action: {action}")

        # Gripper action
        gripper_action = action[0]
        if gripper_action < -self.__gripper_dead_zone:
            self.gripper.close()
        elif gripper_action > self.__gripper_dead_zone:
            self.gripper.open()
        else:
            # No-op for the gripper if action is within the dead zone
            pass

        # Motion
        if self._use_servo:
            linear = action[1:4]
            if self._restrict_position_goal_to_workspace:
                linear = self.restrict_servo_translation_to_workspace(linear)
            if self.__full_3d_orientation:
                angular = action[4:7]
            else:
                angular = [0.0, 0.0, action[4]]
            self.servo(linear=linear, angular=angular)
        else:
            position = self.get_relative_ee_position(action[1:4])
            if self.__full_3d_orientation:
                quat_xyzw = self.get_relative_ee_orientation(
                    rotation=action[4:10], representation="6d"
                )
            else:
                quat_xyzw = self.get_relative_ee_orientation(
                    rotation=action[4], representation="z"
                )
            self.moveit2.move_to_pose(position=position, quat_xyzw=quat_xyzw)
```

It looks too small!!

naa1824, 2025-02-18:
I see

then are you using the default SB3 polices like MlpPolicy or did you created your own?

because I could not find it (I mean the policy code)

AndrejOrsula, 2025-02-18:
All custom policies for the octree observations are located [here](https://github.com/AndrejOrsula/drl_grasping/tree/b2283544d19c3b202b14caa03e233ac4598ccee2/drl_grasping/drl_octree)

AndrejOrsula, 2025-02-18:
For reference, there is also the [Directory Structure](https://github.com/AndrejOrsula/drl_grasping?tab=readme-ov-file#directory-structure) described in README.

[image]