AndrejOrsula/drl_grasping #120: Change the training env

Fuheng188, 2025-03-28:
Sorry to bother you again, but following your suggestion, I successfully ran the Docker version. However, during the training process, I wanted to replace the Moon environment with my own environment while still using your robot. I have tried multiple times but failed each time. No matter how I modify the global status, it always shows as 'error'. Could you please provide me with the specific steps? Once again, thank you for your excellent work

AndrejOrsula, 2025-03-28:
Hello,

If you have the terrain in mind, you can add a new one in the [terrains submodule](https://github.com/AndrejOrsula/drl_grasping/blob/b2283544d19c3b202b14caa03e233ac4598ccee2/drl_grasping/envs/models/terrains/__init__.py) and then use it.

You can also register the task with a default terrain:

https://github.com/AndrejOrsula/drl_grasping/blob/b2283544d19c3b202b14caa03e233ac4598ccee2/drl_grasping/envs/__init__.py#L540

Fuheng188, 2025-03-30:
Thank you for your response. What I mean is that I would like to not only change the terrain but also change the scene. Specifically, I want to try using this end-to-end framework to implement other grasping tasks. Could you provide some modification steps for this? Thank you for your reply.

AndrejOrsula, 2025-03-30:
In that case, you probably want to create a new task [here](https://github.com/AndrejOrsula/drl_grasping/tree/master/drl_grasping/envs/tasks).